McMasterExoskeleton/Electrical #1: Sensors: Design Machine Learning Prototype 2.0

dsouzadakota, 2025-02-28:
Objective:
Create a design for a new prototype with the new PCB breakout boards 

Task:
Create a detailed plan for the new prototype. Plan should include detailed schematics, documentation of wire lengths for the hip, knee and ankle joints, and outline a plan for completion.

Deadline: Friday March 7th

meghang05, 2025-02-28:
Design and schematic are laid out under V2 in the prototypes doc
https://docs.google.com/document/d/1AonTRbTHoHXYbHMtISpS5qf9mnIVo0gSTQg7LomxMOE/edit?tab=t.83affc4oazz

dsouzadakota, 2025-03-04:
Is 95cm the finalized length of the ankle sensor?

meghang05, 2025-03-04:
> [@meghang05](https://github.com/meghang05) Is 95cm the finalized length of the ankle sensor?

Yes, but it hasn't been trimmed yet to that length. 95cm puts us at like mid calf on Juan so we were aiming for the longest length possible without going over a meter

jjllanz, 2025-03-05:
Looks good so far but make sure you're using the design template so as you continue with this prototype it has the correct documentation for info/testing etc.

Johnny-Ash04, 2025-03-14:
So far:
- document has been updated to match the documentation template
- Design is finalized
- Connectors decision is finalized
- Mech details of casing have been finalized and sent to mechanical team 
- Wires for the sensors and multiplexer are soldered to the breakout board
- 6 of 8 new sensors have header pins soldered and their i2c connections have been tested and work (The other 2 are still in packaging but we might not need them if these 6 pass all the test cases)

Current issues and what needs to be done:
- 3 of the 6 multiplexer channels aren't working and need to be fixed (Very likely a soldering issue) 
- Sensors need to be pass test cases and then they can be connected to the prototype "permanently"
- Get the pi and breakout board case from mech so then we can drill 2 more screw holes into breakout board