McMasterExoskeleton/Electrical #2: Sensors: Implementation of Machine Learning Prototype 2.0

dsouzadakota, 2025-02-28:
Objective:
Implement and build the prototype based on completed design.

Task:
- Thoroughly test and validate sensor data
- Test all I2C buses on the multiplexer
- Solder all components to the PCB breakout board and test each ensure to test every component that is soldered to the board (also ensure that the connection is stable for long periods of time)
- Complete details schematics and diagrams of the prototype for mechanical to build the housing
- Ensure to be in contact with the software sensor team to verify how they will be using the prototype to collect data

Deadling: Friday March 7th

Johnny-Ash04, 2025-02-28:
Complete:
- All I2C buses on the multiplexer have been tested and the multiplexer has been soldered to the breakout board
In Progress:
-Soldering other components
-Test sensors and validate data
Next steps:
- Finalize housing/cap dimensions
-

Johnny-Ash04, 2025-03-14:
So far:

document has been updated to match the documentation template
Design is finalized
Connectors decision is finalized
Mech details of casing have been finalized and sent to mechanical team
Wires for the sensors and multiplexer are soldered to the breakout board
6 of 8 new sensors have header pins soldered and their i2c connections have been tested and work (The other 2 are still in packaging but we might not need them if these 6 pass all the test cases)
Current issues and what needs to be done:

3 of the 6 multiplexer channels aren't working and need to be fixed (Very likely a soldering issue)
Sensors need to be pass test cases and then they can be connected to the prototype "permanently"
Get the pi and breakout board case from mech so then we can drill 2 more screw holes into breakout board

meghang05, 2025-03-19:
Progress:

- Prototype has been soldered and tested, with all connections working properly and securely.
- Sensors have been tested separate from the prototype.

To Do:

- Attach sensors using connectors and test connection and data validation.

Johnny-Ash04, 2025-03-31:
Everything is working