McMasterExoskeleton/Electrical #3: Sensors: IMU Data Validation

dsouzadakota, 2025-02-28:
Objective:
Define a series of detailed test cases to validate sensor data and ensure that it is accurate and repeatable. 
Test Case Structure:
- Objective: Clearly state what the test is verifying
- Test Procedure: Define a step-by-step protocol for conducting the test, including test setup and environmental conditions, data collection process, expected variations or edge cases to consider
- Success Criteria: Outline very clear outcome and measurable pass/fail conditions
- Note: The tests that you define should be reproducible. In the future the software team may need to also perform these tests so make sure they are reproducible and properly documented.

Investigate the causes of IMU drift and explore various methods to mitigate it.

Establish a calibration method that corrects any offsets/bias/drift that you have identified in your testing. This calibration procedure will likely be executed upon every boot of the electrical system of the exoskeleton.

meghang05, 2025-02-28:
Test cases have been written and are in progress.

Still to do:
- investigate drift
- condense test cases and identify any that are unnecessary 
- run on all sensors and record
- establish calibration

Johnny-Ash04, 2025-03-31:
All done, changed to only needing Euler orientation data. All test cases passed