McMasterExoskeleton/Electrical #5: Motors: Motor Testing Plan

dsouzadakota, 2025-02-28:
Objective:
Develop a detailed plan for testing the motor in a controlled environment prior to full system integration. 

Task:
The plan should include:
- A test setup to validate torque, speed, and response time.
- Validate the communication between pi and motor is correct with the control hat 
- Tools and software required for validation (R8485 to USB connectors, shielded twisted pair wires)
- Step-by-step instructions for setting up and conducting the tests.

Deadline: Monday March 3rd

jjllanz, 2025-03-05:
please use the design documentation templte for the motor testing plan!

deshpr3, 2025-03-06:
We've combined our test plans into 1 document using the design documentation template: https://mcmasteru365.sharepoint.com/:w:/s/McMasterExoskeleton-Electrical-Motorandactuatorcontrol/EcdYeuLCVu5MgrkgDkJAt4kBWWBg-Ex98te1Jx0C14epGw?e=ZCctCe