McMasterExoskeleton/Electrical #17: Safety: Motor temperature

jjllanz, 2025-03-14:
Motors is going to be doing load testing on the motors week of April 17. We'd like to measure temperatures of the motors during these max loading conditions for safety. The temp sensor themselves should already be tested, so to prepare for this task the main idea is to plan how you will set up the sensors with the motors, and collecting data (very simple script) 

- Plan how you will connect the sensors to the motor
- Simple program to collect data from the temperature sensor

Deadline: Monday march 14
Should be ready to incorporate by the time motors does their testing this week
Please be present at the motor load testing session to set up the temperature sensors

pugno44, 2025-03-18:
Temp sensor will be placed on the flange surrounding the rotating shaft using electrical tape
code will be updated to write to a file and also display a warning if temperature exceeds 40 C (we choose 40 since datasheet said it goes from 0-40 and above 40 is dangerous to touch)

dsouzadakota, 2025-03-22:
Please research the use of thermistors to prevent overtemperature of the motors. Know how they will be connected and how temperature will be monitored. Add a safety section in the doc in the teams motor channel under files/Final Draft/Motor_testing_plan_feedback for how these thermistors will be implemented and a plan to test this.

Once you have an understanding of you the temperature monitoring will work and have completed the safety section put this task back into "To Review" and we will move on to testing this functionality and be able to monitor motor temperature during testing. 

**Deadline Monday 11am**