McMasterExoskeleton/Electrical #18: Motors: Testing PLAN for Motors under load

jjllanz, 2025-03-14:
Develop a comprehensive testing plan to validate motor performance under load and in simulated environmental conditions. Clearly indicate i your documentation that this testing is part of phase II.  Please see your expectations for this test below:

- Mech: Will be providing testing rig to test conditions that mimic the suit’s operational environment on Monday March 17 (as well as a doc with rig information) 
- Elec: Document plan for tests (including tests outlined below), determine testing environment, and tell software what the driver file has to look like to implement these tests (ex intervals, torque) 
- SW members: Develop driver file to set up the motor to move during these conditions
- Safety sensors (Hamza and Emili): Thermal and Endurance Testing: will be monitoring temperature during these tests 
- Safety motors (Abigail): Ensure that these test plans meet safety requirements: https://docs.google.com/document/d/14Or3hwSLP2O9mrKnHKaOZ6eLSBsUQp9v31BFuN2XWXI/edit?tab=t.0#heading=h.ak1i7859aqp2

Tests to include (and additional tests you determine are necessary) 
- Varying Load Conditions: Use a range of load scenarios provided by the mechanical team to simulate different weight and resistance conditions.
- Electrical Measurements Record voltage and current during various load scenarios to monitor performance and detect potential issues.
- Simultaneous Motor Operation Test multiple motors operating together to assess overall system behavior under load.
- Environmental Simulation Testing:  Set up a simulation environment to replicate real-world distances and conditions (e.g., battery distances, wiring lengths) to ensure that the test conditions closely mimic the actual exoskeleton setup.
- Movement Under Load: Test motor behavior during simulated movement, such as starting, stopping, and continuous operation under load, to verify that performance remains consistent during real-world operation.
- Safety and Emergency Response Testing:  Evaluate how the system responds to abrupt load changes or emergency stops. Ensure that safety protocols are triggered appropriately and that the system maintains safe operational limits.

**Outcome:**  A clear and concise test plan with sensor inputs to drive the motor control. Add to document called Exoskeleton Motors: Integrated Testing following template.
**Deadline:** March 17 2025 11:59 PM
**NOTE:** if the script from SW isn't ready by monday night, use MotionLab to implement these tests. The motor test plan MUST be completed by Monday night. The script is a helpful building block for completing the driver API, so the sw team should continue to work on this. Once it's complete test it with the motors to validate program success.

!!!Please move the issue into “In Review” after completing the testing plans to allow us to review the plan before you begin the actual tests (a separate sub-issue). Once the plan is in Done, proceed with implementation!!!

tphilelin, 2025-03-18:
[Exoskeleton Motors Integrated Testing.docx](https://github.com/user-attachments/files/19304964/Exoskeleton.Motors.Integrated.Testing.docx)

The motor testing plan is attached above and can also be found in the teams Electrical - Motor Control and actuator -> Files -> Final Draft -> Exoskeleton Motors Integrated Testing.docx

dsouzadakota, 2025-03-20:
Please update the plan with feedback from Dr. Bilgin!

tphilelin, 2025-03-20:
where can we find the feedback?

jjllanz, 2025-03-20:
> Please update the plan with feedback from Dr. Bilgin!

 "Motor_testing_feedback" into exoskeleton teams channel -> motors -> files
[https://mcmasteru365.sharepoint.com/:w:/r/sites/McMasterExoskeleton-Electrical-Motorandactuatorcontrol/Shared%20Documents/Electrical%20-%20Motor%20and%20actuator%20control/Final%20Draft/[Motor_testing_feedback.docx](https://mcmasteru365.sharepoint.com/:w:/r/sites/McMasterExoskeleton-Electrical-Motorandactuatorcontrol/Shared%20Documents/Electrical%20-%20Motor%20and%20actuator%20control/Final%20Draft/Motor_testing_feedback.docx?d=wc9b0b802297742229423d72b26daef6b&csf=1&web=1&e=dW90FO)?d=wc9b0b802297742229423d72b26daef6b&csf=1&web=1&e=dW90FO](url)

dsouzadakota, 2025-03-25:
Feed back should be implemented by **March 26th 11:59pm**

tphilelin, 2025-03-27:
The updated motor testing plan can be located in the teams under Motor and Actuator Control -> Files -> Final Draft -> Motor_testing_feedback.docx 

NOTE: Ritvik and Cameron are meeting tomorrow to figure out more about what would be the ideal loads that would simulate the motor best for the exoskeleton so the load tests (not including the lever arm dyno) still need to be ironed out a little more.

closed 2025-03-29 by jjllanz